unmannedlab/imu_lidar_calibration #4: Not enough IMU excitation, below threshold

Wuxinxiaoshifu, 2021-12-15:
[InertialInitializer::initialize_with_imu] Not enough IMU excitation, below threshold 0.0241 < 0.2500
[ INFO] [1639578655.676617553]: Lidar scan no: 34
[ INFO] [1639578655.676645568]: No of lodom: 0
[ INFO] [1639578655.758270962]: No of imu measurements: 40
Failed to find match for field 'intensity'.
Failed to find match for field 'ring'.
Failed to find match for field 't'.
[InertialInitializer::initialize_with_imu] Not enough IMU excitation, below threshold 0.0476 < 0.2500
[ INFO] [1639578655.760411787]: Lidar scan no: 35
[ INFO] [1639578655.760441037]: No of lodom: 0
[ INFO] [1639578655.844012505]: No of imu measurements: 40
Failed to find match for field 'intensity'.
Failed to find match for field 'ring'.
Failed to find match for field 't'.
[InertialInitializer::initialize_with_imu] Initialization done!!!!
terminate called after throwing an instance of 'pcl::IsNotDenseException'
  what():  : Can't use 2D indexing with a unorganized point cloud
================================================================================REQUIRED process [ros_test_node-1] has died!

SubMishMar, 2021-12-15:
Looks like the issue is with the pointcloud, not with IMU readings.

Wuxinxiaoshifu, 2021-12-16:
> Looks like the issue is with the pointcloud, not with IMU readings.

thanks,it is the timestamp which camera is not same with as imu

Wuxinxiaoshifu, 2021-12-16:
> Looks like the issue is with the pointcloud, not with IMU readings.

thanks,but i changed the timestamp,it doesn't work.

Wuxinxiaoshifu, 2021-12-16:
> 
i changed lidar's stamp to synchroize with imu's stamp,but it doesn't work;
can you tell me what is init_imu_thresh and init_window_time?
(/// If the above standard dev is below a threshold and we are waiting for a "jerk" then we want to wait till it is exceeded)

zfc-zfc, 2021-12-23:
> > 
> 
> i changed lidar's stamp to synchroize with imu's stamp,but it doesn't work; can you tell me what is init_imu_thresh and init_window_time? (/// If the above standard dev is below a threshold and we are waiting for a "jerk" then we want to wait till it is exceeded)

This is caused by unorganized pointcloud (the height is 1). You can modify the bag reader module to support unorganized pointcloud.

Wuxinxiaoshifu, 2021-12-26:
> This is caused by unorganized pointcloud (the height is 1). You can modify the bag reader module to support unorganized pointcloud.

thanks

huanghqc, 2022-07-05:
> > > > 
> > > 
> > > 
> > > 我更改了激光雷达的图章以与imu的图章同步，但它不起作用;你能告诉我什么是init_imu_thresh和init_window_time吗？（/// 如果上述标准 dev 低于阈值，并且我们正在等待“混蛋”，那么我们希望等到超过它）
> > 
> > 
> > 这是由无组织的点云（高度为1）引起的。您可以修改袋子阅读器模块以支持无组织的点云。
> 
> 谢谢

请问您是如何修改的，能否详细说明一下，多谢

xzh313, 2022-11-15:
> > > > > 
> > > > 
> > > > 
> > > > 我更改了激光雷达的图章以与imu的图章同步，但它不起作用;你能告诉我什么是init_imu_thresh和init_window_time吗？（/// 如果上述标准 dev 低于阈值，并且我们正在等待“混蛋”，那么我们希望等到超过它）
> > > 
> > > 
> > > 这是由无组织的点云（高度为1）引起的。您可以修改袋子阅读器模块以支持无组织的点云。
> > 
> > 
> > 谢谢
> 
> 请问您是如何修改的，能否详细说明一下，多谢
请问下最后解决了吗？

xzh313, 2022-11-15:
> > > > > 
> > > > 
> > > > 
> > > > 我更改了激光雷达的图章以与imu的图章同步，但它不起作用;你能告诉我什么是init_imu_thresh和init_window_time吗？（/// 如果上述标准 dev 低于阈值，并且我们正在等待“混蛋”，那么我们希望等到超过它）
> > > 
> > > 
> > > 这是由无组织的点云（高度为1）引起的。您可以修改袋子阅读器模块以支持无组织的点云。
> > 
> > 
> > 谢谢
> 
> 请问您是如何修改的，能否详细说明一下，多谢

请问最后解决了吗？ @huanghqc

HyeonnSeo, 2023-05-24:
> This is caused by unorganized pointcloud (the height is 1). You can modify the bag reader module to support unorganized pointcloud.

what it means? @zfc-zfc