unmannedlab/imu_lidar_calibration #5: Wrong calibration result with my own data.

zfc-zfc, 2021-12-23:
Hi, I tried your method with my data collected using Hesai PandarXT(32 line) and an external IMU. The data is synchronized, but the calibration result is wrong (The ground truth should be (-0.12, 0, 0.15), (0,0,-180) ), shown as follows:

Time taken for deskewing: 2.45107 [s]
[Propagation] successfully propagated
[Update] successful updateScan2Scan 
[Update] successful updateScan2GlobalMap
[ INFO] [1640251233.514128561]: Lidar scan no: 557
[ INFO] [1640251233.514159010]: No of lodom: 498
[ INFO] [1640251233.514329557]: Reached end of bag
[ INFO] [1640251233.514342060]: Kalman Filtering took : 1295.45 [s]
Writing KF calibration result to: /home/dji/workspace/Texas_LICalib_ws/src/imu_lidar_calibration/linkalibr/data/I_T_L_final.txt
[ INFO] [1640251233.514537621]: Translation [in m]: -0.113965 -0.195481   -1.0932
[ INFO] [1640251233.514552354]: Euler Angles [in deg]:  174.288   179.51 -178.068

And I tested the same bag twice, the result are quite different. Could you please tell me the reason? Thx

SubMishMar, 2021-12-23:
May be reading https://arxiv.org/abs/2104.12280 will provide more insights.

Generic advice will be calibrate the IMU intrinsics, and most importantly exhibit motion which sufficiently excites all DoFs while collecting calibration data. I have not tested this particular LiDAR but I am not sure why scan deskewing is taking 2.5 seconds even though it's a 32 channel lidar.

zfc-zfc, 2021-12-24:
> 

Thanks, I have carefully read your paper and I cited your good work in my paper. Maybe I can send you the test data which might be helpful to find out the reason of large time consumption of scan deskwing. Since lots of LiDARs provide unorganized pointcloud (the height is always 1), it would be great if your code support unorganized input pointcloud. Thank you so much.

liang0724s, 2022-01-06:
@zfc-zfc Hi ,
i met a building error #7 ,it would be great if you tell me your building progress. Thank you so much.

liang0724s, 2022-01-11:
@zfc-zfc Hi, 
did you meet the node(ros_calib_init) died when run the command "roslaunch linkalibr ros_calib_init.launch"?

started roslaunch server http://minzhus:45557/
SUMMARY
========

CLEAR PARAMETERS
 * /ros_calib_init/
 * /ros_calib_init_optimizer/

PARAMETERS
 * /ros_calib_init/bag_durr: -1
 * /ros_calib_init/bag_start: 0.0
 * /ros_calib_init/ndt_resolution: 0.25
 * /ros_calib_init/path_bag: /home/minzhus/rec...
 * /ros_calib_init/topic_imu: /vectornav/IMU
 * /ros_calib_init/topic_lidar: /os_cloud_node/po...
 * /ros_calib_init_optimizer/accelerometer_noise_density: 0.094666
 * /ros_calib_init_optimizer/calibration_result_filename: /home/minzhus/rec...
 * /ros_calib_init_optimizer/gyroscope_noise_density: 6.243e-05
 * /ros_calib_init_optimizer/max_frames: 550
 * /rosdistro: kinetic
 * /rosversion: 1.12.17

NODES
  /
    ros_calib_init (linkalibr/ros_calib_init)
    ros_calib_init_optimizer (linkalibr/ros_calib_init_optimizer)

ROS_MASTER_URI=http://localhost:11311

WARNING: Package name "imuPacket" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits, underscores, and dashes.
process[ros_calib_init-1]: started with pid [21421]
process[ros_calib_init_optimizer-2]: started with pid [21422]
[ INFO] [1641893388.759679507]: Loaded accelerometer_noise_density: 0.094666
[ INFO] [1641893388.759999244]: Loaded gyroscope_noise_density: 6.243e-05
[ INFO] [1641893388.760289321]: Loaded max_frames: 550
[ INFO] [1641893388.760535149]: Loaded calibration_result_filename: /home/minzhus/receive/Lidar_imu_calib/TAMU/catkin_ws_ov/src/imu_lidar_calibration/linkalibr/data/I_T_L_init.txt
[ WARN] [1641893388.760557122]: [calibInitOptimizer::calibInitOptimizer] calibInitOptimizer constructor
**================================================================================REQUIRED process [ros_calib_init-1] has died!
process has died [pid 21421, exit code -11, cmd /home/minzhus/receive/Lidar_imu_calib/TAMU/catkin_ws_ov/devel/lib/linkalibr/ros_calib_init __name:=ros_calib_init __log:=/home/minzhus/.ros/log/d051f658-72ba-11ec-a667-6609fd46ba79/ros_calib_init-1.log].
log file: /home/minzhus/.ros/log/d051f658-72ba-11ec-a667-6609fd46ba79/ros_calib_init-1*.log
Initiating shutdown!
================================================================================**
[ros_calib_init_optimizer-2] killing on exit
[ros_calib_init-1] killing on exit