unmannedlab/imu_lidar_calibration #8: In your paper, how do you get formula 3 from formula 2？

772894624, 2022-03-01:
Hi~
First of all, thank you for your contribution.
When I read your paper, I didn't understand how to get formula 3 from formula 2. Can you tell me? thanks!

SubMishMar, 2022-03-02:
Oh there's a missing citation, but it's fairly standard. You may check the 5th page of this paper: https://journals.sagepub.com/doi/pdf/10.1177/027836499501400301

772894624, 2022-03-03:
Thanks，It solves my doubts. 
When I use my own sensor(VLP16 LIDAR) for calibration, the program prompts the following information:

    [ INFO] [1646294431.779268199]: bag start: 0
    [ INFO] [1646294431.779283115]: bag duration: -1
    [ INFO] [1646294431.834946031]: Time start = 1.64616e+09
    [ INFO] [1646294431.834979136]: Time end = 1.64616e+09
    Propagator::select_IMU_readings(): No IMU measurements.
    Propagator::select_imu_readings(): No IMU measurement to propgate with (0 of 2) 
    Propagator::select_imu_readings(): No IMU measurements to propagate with (1 of 2).
    [ INFO] [1646294431.840939246]: No of imu measurements: 6
    Failed to find match for field 'ring'.
    Failed to find match for field 't'.
    [InertialInitializer::initialize_with_imu] One of the windows is empty, not enough readings 
    [ INFO] [1646294431.841472401]: Lidar scan no: 0
    [ INFO] [1646294431.841505614]: No of lodom: 0
    [ INFO] [1646294431.842859383]: No of imu measurements: 41
    Failed to find match for field 'ring'.
    Failed to find match for field 't'.
    [InertialInitializer::initialize_with_imu] One of the windows is empty, not enough readings 

What causes this problem? How to solve this problem? thank you!

xzh313, 2022-11-15:
> Thanks，It solves my doubts. When I use my own sensor(VLP16 LIDAR) for calibration, the program prompts the following information:
>
> ```
> [ INFO] [1646294431.779268199]: bag start: 0
> [ INFO] [...]

have you solved this problem?I meet this problem too

772894624, 2022-11-18:
I I didn't solve the problem. Because I use VLP16, I use other algorithms.@xzh313

xzh313, 2022-11-18:
Which algorithms did you use? Can you share a link to github? How about your final result for your matrix？THANKS!