unmannedlab/imu_lidar_calibration #11: REQUIRED process [ros_calib_init_optimizer-2] has died!

huanghqc, 2022-05-09:
运行roslaunch linkalibr ros_calib_init.launch后报错如下：

[ INFO] [1652091019.397516439]: Frame: 547 / 550
[ INFO] [1652091019.652656943]: Frame: 548 / 550
[ INFO] [1652091020.011336717]: Frame: 549 / 550
[ INFO] [1652091020.301443733]: Frame: 550 / 550
Rot3: 

   0.999763   0.0107452   0.0189149
  0.0106649   -0.999934  0.00433886
  0.0189603 -0.00413611   -0.999812
Euler Angles: 0.248644  178.916  179.384
Marginal Covariance
   0.596503 -0.00196232 -0.00136384
-0.00196232    0.587782   0.0279263
-0.00136384   0.0279263    0.594626
================================================================================REQUIRED process [ros_calib_init_optimizer-2] has died!
process has finished cleanly
log file: /home/hqc/.ros/log/3c887e46-cf4f-11ec-aca0-9822ef00c7d8/ros_calib_init_optimizer-2*.log
Initiating shutdown!
================================================================================
[ros_calib_init_optimizer-2] killing on exit
[ros_calib_init-1] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

huanghqc, 2022-07-04:
@SubMishMar I didn't solve it, but I don't think this error will have much impact on the result, so I haven't considered this error

alwynmathew, 2023-08-15:
The process died on https://github.com/unmannedlab/imu_lidar_calibration/blob/main/linkalibr/src/ros_calib_init_optimizer.cpp#L173 just after processing the last frame.
```
[ INFO] [1692115069.770867042]: Frame: 550 / 550
================================================================================
REQUIRED process [ros_calib_init_optimizer-2] has died!
process has died [pid 1091109, exit code -11, cmd /home/alwyn/catkin_ws2/devel/lib/linkalibr/ros_calib_init_optimizer __name:=ros_calib_init_optimizer __log:=/home/alwyn/.ros/log/b643b8ce-36d0-11ee-90ab-2fb04df3755f/ros_calib_init_optimizer-2.log].
log file: /home/alwyn/.ros/log/b643b8ce-36d0-11ee-90ab-2fb04df3755f/ros_calib_init_optimizer-2*.log
Initiating shutdown!
================================================================================
[ros_calib_init_optimizer-2] killing on exit
[ros_calib_init-1] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
```