unmannedlab/imu_lidar_calibration #13: Could I use this tool on the Ground robot dataset?

Taeyoung96, 2022-06-24:
@SubMishMar   
Thanks for your great work.  

I have a question.  

I want to know the extrinsic calibration of Velodyne VLP 16 and Bosch BMI055(built-in IMU on realsense D435i).  

Since it is attached to the ground robot, so it is difficult to obtain the same motion as the [Data collection procedure video](https://youtu.be/2IX5LVTDkLc).  

Can I use this tool to get an extrinsic parameter with the dataset I got from moving the Ground robot?  

Thanks,

Taeyoung96, 2022-07-04:
@SubMishMar   
It's attached to the husky robot, and I couldn't show the robot's setup right now. :cry:    
Thanks for your help.  

BTW, When I run the `roslaunch linkalibr linkalibr_ouster_vectornav.launch` with my own dataset.  
I got this error.  
```
SUMMARY
========

CLEAR PARAMETERS
 * /ros_test_node/

PARAMETERS
 * /ros_test_node/accelerometer_noise_density: 0.1
 * /ros_test_node/accelerometer_random_walk: 0.001
 * /ros_test_node/bag_durr: -1
 * /ros_test_node/bag_start: 0.0
 * /ros_test_node/calib_extrinsics: True
 * /ros_test_node/calib_lidar_timeoffset: False
 * /ros_test_node/calib_lidarimu_dt: 0.0
 * /ros_test_node/calibration_result_velodyne_filename: /home/catkin_ws/s...
 * /ros_test_node/deskewed_scan_folder_name: /home/catkin_ws/s...
 * /ros_test_node/do_planar_update: False
 * /ros_test_node/do_undistortion: True
 * /ros_test_node/downsample_for_mapping: True
 * /ros_test_node/gen_data_for_optimization: False
 * /ros_test_node/gravity: [0.0, 0.0, 9.81]
 * /ros_test_node/gyroscope_noise_density: 0.01
 * /ros_test_node/gyroscope_random_walk: 2e-05
 * /ros_test_node/imu_measurements_filename: /home/catkin_ws/s...
 * /ros_test_node/imu_trajectory_filename: /home/catkin_ws/s...
 * /ros_test_node/init_calibration_result_velodyne_filename: /home/catkin_ws/s...
 * /ros_test_node/init_imu_thresh: 0.25
 * /ros_test_node/init_window_time: 0.5
 * /ros_test_node/lidar_trajectory_filename: /home/catkin_ws/s...
 * /ros_test_node/limit_map_size: True
 * /ros_test_node/lin_bias_filename: /home/catkin_ws/s...
 * /ros_test_node/lin_calib_dt_filename: /home/catkin_ws/s...
 * /ros_test_node/lin_calib_extrinsic_filename: /home/catkin_ws/s...
 * /ros_test_node/lin_trajectory_filename: /home/catkin_ws/s...
 * /ros_test_node/lin_velocity_filename: /home/catkin_ws/s...
 * /ros_test_node/lo_trajectory_filename: /home/catkin_ws/s...
 * /ros_test_node/max_clones: 2
 * /ros_test_node/ndt_resolution: 0.25
 * /ros_test_node/no_of_scans_for_map: 200
 * /ros_test_node/path_bag: /home/catkin_ws/s...
 * /ros_test_node/planar_points_deskewed_map_filename: /home/catkin_ws/s...
 * /ros_test_node/planar_points_raw_scans_filename: /home/catkin_ws/s...
 * /ros_test_node/plane_params_filename: /home/catkin_ws/s...
 * /ros_test_node/raw_scan_folder_name: /home/catkin_ws/s...
 * /ros_test_node/state_init_rx_noise: 0.1
 * /ros_test_node/state_init_ry_noise: 0.1
 * /ros_test_node/state_init_rz_noise: 0.1
 * /ros_test_node/state_init_timeoffset_noise: 0.01
 * /ros_test_node/state_init_x_noise: 0.5
 * /ros_test_node/state_init_y_noise: 0.5
 * /ros_test_node/state_init_z_noise: 0.5
 * /ros_test_node/surfel_map_filename: /home/catkin_ws/s...
 * /ros_test_node/topic_imu: /camera/imu
 * /ros_test_node/topic_lidar: /velodyne_points
 * /ros_test_node/updater_chi2_multiplier: 1
 * /ros_test_node/updater_do_chi2_check: False
 * /ros_test_node/updater_rotation_noise: 0.1
 * /ros_test_node/updater_translation_noise: 0.1
 * /ros_test_node/use_fej: False
 * /ros_test_node/use_imuavg: True
 * /ros_test_node/use_rk4int: True
 * /ros_test_node/view_surfel_map: True
 * /rosdistro: melodic
[... 202 lines cut ...]
[InertialInitializer::initialize_with_imu] Not enough IMU excitation, below threshold 0.0007 < 0.2500
[ INFO] [1656905316.460330329]: Input point cloud size : 0
[ INFO] [1656905316.460355809]: Lidar scan no: 24
[ INFO] [1656905316.460484992]: No of lodom: 0
[ INFO] [1656905316.510322219]: No of imu measurements: 40
[InertialInitializer::initialize_with_imu] Initialization done!!!!
terminate called after throwing an instance of 'pcl::IsNotDenseException'
  what():  : Can't use 2D indexing with a unorganized point cloud
================================================================================REQUIRED process [ros_test_node-1] has died!
process has died [pid 289, exit code -6, cmd /home/catkin_ws/devel/lib/linkalibr/ros_test_node __name:=ros_test_node __log:=/root/.ros/log/e4b51258-f792-11ec-b3fc-3497f6927eaa/ros_test_node-1.log].
log file: /root/.ros/log/e4b51258-f792-11ec-b3fc-3497f6927eaa/ros_test_node-1*.log
Initiating shutdown!
================================================================================
[ros_test_node-1] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
```

I put in the initial value `I_T_L_init.txt` and I matched all the paths with my custom dataset and executed it.  
I use Velodyne VLP 16 and Bosch BMI055(built-in IMU on realsense D435i).  
The problem is I couldn't get LiDAR odometry with Velodyne VLP 16.  
I'd appreciate your help.  

Thanks,

huanghqc, 2022-07-04:
@Taeyoung96   I encountered the same problem. How did you solve it?

Taeyoung96, 2022-07-04:
@huanghqc  
Hi, I haven't solved it yet. :cry:

huanghqc, 2022-07-19:
> @huanghqc 嗨，我还没有解决它。😢

@Taeyoung96 你解决了吗

Taeyoung96, 2022-07-19:
@huanghqc I didn't solve it, so I use another [tool](https://github.com/chennuo0125-HIT/lidar_imu_calib)

huanghqc, 2022-07-19:
> @huanghqc我没有解决它，所以我使用另一个[工具](https://github.com/chennuo0125-HIT/lidar_imu_calib)

这个工具好像只能标定旋转矩阵，并不能标定平移矩阵

Taeyoung96, 2022-07-19:
@huanghqc   
Yes, You are right. The translation vector has not been found accurately yet. :cry:

arjunskumar, 2022-08-16:
@Taeyoung96 have you able to find translation vector?

Taeyoung96, 2022-08-16:
@arjunskumar Unfortunately, In the case of the translation vector, I got it by manual.  
I looked directly at the sensor attached to the robot and calculated for it.

arjunskumar, 2022-08-16:
@Taeyoung96 thanks, for the rotation matrix, you used which package?

Taeyoung96, 2022-08-16:
@arjunskumar I mentioned above, I use [this tool](https://github.com/chennuo0125-HIT/lidar_imu_calib)! :smile:

narayanlongani, 2023-02-10:
> @SubMishMar It's attached to the husky robot, and I couldn't show the robot's setup right now.cry Thanks for your help.
>
> BTW, When I run the `roslaunch linkalibr linkalibr_ouster_vectornav.launch` [...]

I am having similar problem with Velodyne 16

If you have any solution... please share

[ INFO] [1676007320.087536270]: Time start = 1.67584e+09
[ INFO] [1676007320.087550987]: Time end = 1.67584e+09
Propagator::select_IMU_readings(): No IMU measurements.
Propagator::select_imu_readings(): No IMU measurement to propgate with (0 of 2) 
[ INFO] [1676007320.088277752]: No of imu measurements: 2
Failed to find match for field 't'.
[InertialInitializer::initialize_with_imu] One of the windows is empty, not enough readings 
[ INFO] [1676007320.088748826]: Lidar scan no: 0
[ INFO] [1676007320.088770116]: No of lodom: 0
Propagator::select_imu_readings(): No IMU measurements to propagate with (1 of 2).
[ INFO] [1676007320.089149218]: No of imu measurements: 11
Failed to find match for field 't'.
[InertialInitializer::initialize_with_imu] One of the windows is empty, not enough readings 
[ INFO] [1676007320.089447328]: Lidar scan no: 1
[ INFO] [1676007320.089459882]: No of lodom: 0
[ INFO] [1676007320.089643226]: No of imu measurements: 11
Failed to find match for field 't'.
[InertialInitializer::initialize_with_imu] One of the windows is empty, not enough readings 
[ INFO] [1676007320.089944301]: Lidar scan no: 2
[ INFO] [1676007320.089951645]: No of lodom: 0
[ INFO] [1676007320.090369560]: No of imu measurements: 9
Failed to find match for field 't'.
[InertialInitializer::initialize_with_imu] One of the windows is empty, not enough readings 
[ INFO] [1676007320.090645358]: Lidar scan no: 3
[ INFO] [1676007320.090653222]: No of lodom: 0
[ INFO] [1676007320.091115180]: No of imu measurements: 9
Failed to find match for field 't'.
[InertialInitializer::initialize_with_imu] One of the windows is empty, not enough readings 
[ INFO] [1676007320.091388473]: Lidar scan no: 4
[ INFO] [1676007320.091396218]: No of lodom: 0
[ INFO] [1676007320.091837927]: No of imu measurements: 10
Failed to find match for field 't'.
[InertialInitializer::initialize_with_imu] Not enough IMU excitation, below threshold 0.0021 < 0.2000
[ INFO] [1676007320.092113354]: Lidar scan no: 5
[ INFO] [1676007320.092120959]: No of lodom: 0
[ INFO] [1676007320.092796046]: No of imu measurements: 12
Failed to find match for field 't'.
[InertialInitializer::initialize_with_imu] Not enough IMU excitation, below threshold 0.0022 < 0.2000
[ INFO] [1676007320.093071233]: Lidar scan no: 6
[ INFO] [1676007320.093082003]: No of lodom: 0
[ INFO] [1676007320.093734378]: No of imu measurements: 9
Failed to find match for field 't'.
[InertialInitializer::initialize_with_imu] Not enough IMU excitation, below threshold 0.0022 < 0.2000
[ INFO] [1676007320.094170527]: Lidar scan no: 7
[ INFO] [1676007320.094179745]: No of lodom: 0
[ INFO] [1676007320.094768049]: No of imu measurements: 10
Failed to find match for field 't'.
[InertialInitializer::initialize_with_imu] Not enough IMU excitation, below threshold 0.0020 < 0.2000
[ INFO] [1676007320.094987031]: Lidar scan no: 8
[ INFO] [1676007320.094994214]: No of lodom: 0
[ INFO] [1676007320.095711130]: No of imu measurements: 10
Failed to find match for field 't'.
[InertialInitializer::initialize_with_imu] Not enough IMU excitation, below threshold 0.0018 < 0.2000
[ INFO] [1676007320.095928108]: Lidar scan no: 9
[ INFO] [1676007320.095935121]: No of lodom: 0
[ INFO] [1676007320.096769638]: No of imu measurements: 10
Failed to find match for field 't'.
[InertialInitializer::initialize_with_imu] Not enough IMU excitation, below threshold 0.0017 < 0.2000
[ INFO] [1676007320.096986665]: Lidar scan no: 10
[ INFO] [1676007320.096993779]: No of lodom: 0
[ INFO] [1676007320.097789012]: No of imu measurements: 10
Failed to find match for field 't'.
[InertialInitializer::initialize_with_imu] Not enough IMU excitation, below threshold 0.0017 < 0.2000
[ INFO] [1676007320.098006560]: Lidar scan no: 11
[ INFO] [1676007320.098013153]: No of lodom: 0
[ INFO] [1676007320.098940323]: No of imu measurements: 10
Failed to find match for field 't'.
[InertialInitializer::initialize_with_imu] Not enough IMU excitation, below threshold 0.0095 < 0.2000
[ INFO] [1676007320.099154866]: Lidar scan no: 12
[ INFO] [1676007320.099161529]: No of lodom: 0
[ INFO] [1676007320.100221549]: No of imu measurements: 10
Failed to find match for field 't'.
[InertialInitializer::initialize_with_imu] Initialization done!!!!
terminate called after throwing an instance of 'pcl::IsNotDenseException'
  what():  : Can't use 2D indexing with a unorganized point cloud
================================================================================REQUIRED process [ros_test_node-1] has died!
process has died [pid 11649, exit code -6, cmd /home/a/target_free_imu_lidar/devel/lib/linkalibr/ros_test_node __name:=ros_test_node __log:=/home/a/.ros/log/cf850076-a8f6-11ed-80d5-88aedd1681ad/ros_test_node-1.log].
log file: /home/a/.ros/log/cf850076-a8f6-11ed-80d5-88aedd1681ad/ros_test_node-1*.log
Initiating shutdown!
================================================================================
[ros_test_node-1] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done