unmannedlab/imu_lidar_calibration #15: Launch error

SyahirMuzni, 2022-10-17:
Hi,
I have successfully built the package and did the intrinsic calibration. I did change some of the requirements in the launch file to match up with my sensors.
But I have an error when I tried to launch the package. I did catkin_make clean & change the permissions as well. But there's no difference.

*ERROR: cannot launch node of type [linkalibr/ros_calib_init]: Cannot locate node of type [ros_calib_init] in package [linkalibr]. Make sure file exists in package path and permission is set to executable (chmod +x)
ERROR: cannot launch node of type [linkalibr/ros_calib_init_optimizer]: Cannot locate node of type [ros_calib_init_optimizer] in package [linkalibr]. Make sure file exists in package path and permission is set to executable (chmod +x)*

SyahirMuzni, 2022-10-17:
For your information, 
My system is running ubuntu 20.04 (noetic)

SubMishMar, 2022-10-17:
Have you sourced the workspace? 

To source: Run
`
source devel/setup.bash`
at the top level of your workspace. http://wiki.ros.org/catkin/Tutorials/create_a_workspace

I have never tried running this on ros noetic.

SyahirMuzni, 2022-10-17:
Yes I have sourced it.

SubMishMar, 2022-10-19:
Okay. Then I have no idea. I would suggest you to use the docker version. You can find it in the README.

songWell, 2022-11-22:
Hi, thank you for your work
CMakeLists.txt is changed

`commit 56e7b6ae7042b0d476eb1dffcddea6be9fbcad4b
Author: SubMishMar <subodheenitr@gmail.com>
Date:   Thu Jul 7 08:39:13 2022 -0500

    backing up

diff --git a/linkalibr/CMakeLists.txt b/linkalibr/CMakeLists.txt
index 059052d..436a471 100644
--- a/linkalibr/CMakeLists.txt
+++ b/linkalibr/CMakeLists.txt
@@ -31,7 +31,7 @@ find_package(Eigen3 REQUIRED)
 find_package(Boost REQUIRED COMPONENTS system filesystem thread date_time)
 #find_package(PCL REQUIRED)
 #find_package(Ceres REQUIRED)
-find_package(GTSAM REQUIRED)
+#find_package(GTSAM REQUIRED)
 
 #link_directories(${PCL_LIBRARY_DIRS})
 #add_definitions(${PCL_DEFINITIONS})
@@ -60,7 +60,7 @@ include_directories(
         ${catkin_INCLUDE_DIRS}
 #        ${PCL_INCLUDE_DIRS}
 #        ${CERES_INCLUDE_DIRS}
-        ${GTSAM_INCLUDE_DIR}
+#        ${GTSAM_INCLUDE_DIR}
 )
 
 # Set link libraries used by all binaries
@@ -69,7 +69,7 @@ list(APPEND thirdparty_libraries
         ${catkin_LIBRARIES}
         ${PCL_LIBRARIES}
         ${CERES_LIBRARIES}
-                gtsam
+        #        gtsam
         )
 
 ##################################################
@@ -93,24 +93,6 @@ target_include_directories(linkalibr_lib PUBLIC src)
 add_executable(ros_test_node src/ros_linkalibr_test.cpp)
 target_link_libraries(ros_test_node linkalibr_lib ${catkin_LIBRARIES})
 
-add_executable(ros_calib_init src/ros_calib_init.cpp)
-target_link_libraries(ros_calib_init linkalibr_lib ${catkin_LIBRARIES})
-#
-#add_executable(ros_calib_init_optimizer src/ros_calib_init_optimizer.cpp)
-#target_link_libraries(ros_calib_init_optimizer
-#        gtsam
-##        ${CERES_LIBRARIES}
-#        ${catkin_LIBRARIES})
-
-#add_executable(ros_batchoptimizer src/ros_batchoptimization.cpp
-#                                  src/batch_optimization/LoadPriorKnowledge.cpp
-#                                  src/batch_optimization/BatchOptimizer.cpp
-#                                  src/batch_optimization/PointToPlaneFactor.cpp
-#                                  src/batch_optimization/PointToPlaneFactor2.cpp
-#                                  src/batch_optimization/PointToPlaneFactor3.cpp)
-#
-#target_link_libraries(ros_batchoptimizer gtsam ${catkin_LIBRARIES})
`

alwynmathew, 2023-08-15:
Replace [linkalibr/CMakeLists.txt](https://github.com/unmannedlab/imu_lidar_calibration/blob/main/linkalibr/CMakeLists.txt) with the below
```
cmake_minimum_required(VERSION 3.10)
# Project name
project(linkalibr)
set(CMAKE_CXX_STANDARD 14)
set(CMAKE_BUILD_TYPE "RELEASE")
#SET(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -march=native ")
#set(CMAKE_CXX_FLAGS_DEBUG "${CMAKE_CXX_FLAGS_DEBUG} -O1")
find_package(OpenMP)
if (OPENMP_FOUND)
    set (CMAKE_C_FLAGS "${CMAKE_C_FLAGS} ${OpenMP_C_FLAGS}")
    set (CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} ${OpenMP_CXX_FLAGS}")
endif()

# Find catkin (the ROS build system)
find_package(catkin QUIET COMPONENTS
        roscpp
        rosbag
        std_msgs
        geometry_msgs
        sensor_msgs
        nav_msgs
        ndt_omp
        tf
        imuPacket
        pcl_ros
        pcl_conversions
)

# Include libraries
find_package(Eigen3 REQUIRED)
find_package(Boost REQUIRED COMPONENTS system filesystem thread date_time)
#find_package(PCL REQUIRED)
#find_package(Ceres REQUIRED)
#find_package(GTSAM REQUIRED)

#link_directories(${PCL_LIBRARY_DIRS})
#add_definitions(${PCL_DEFINITIONS})

# display message to user
message(STATUS "EIGEN VERSION: " ${EIGEN3_VERSION})
message(STATUS "BOOST VERSION: " ${Boost_VERSION})

# Describe catkin project
if (catkin_FOUND)
    add_definitions(-DROS_AVAILABLE=1)
    catkin_package(
            CATKIN_DEPENDS roscpp rosbag tf std_msgs geometry_msgs sensor_msgs nav_msgs imuPacket
            INCLUDE_DIRS src
            LIBRARIES linkalibr_lib
    )
else()
    message(WARNING "CATKIN NOT FOUND BUILDING WITHOUT ROS!")
endif()

# Include our header files
include_directories(
        src
        ${EIGEN3_INCLUDE_DIR}
        ${Boost_INCLUDE_DIRS}
        ${catkin_INCLUDE_DIRS}
#        ${PCL_INCLUDE_DIRS}
#        ${CERES_INCLUDE_DIRS}
#        ${GTSAM_INCLUDE_DIR}
)

# Set link libraries used by all binaries
list(APPEND thirdparty_libraries
        ${Boost_LIBRARIES}
        ${catkin_LIBRARIES}
        ${PCL_LIBRARIES}
        ${CERES_LIBRARIES}
        #        gtsam
        )

##################################################
# Make the core library
##################################################
add_library(linkalibr_lib SHARED
        src/init/InertialInitializer.cpp
        src/state/State.cpp
        src/state/StateHelper.cpp
        src/state/Propagator.cpp
        src/update/UpdaterLidarOdometry.cpp
        src/track/lidarOdometry.cpp
        src/core/lincalibManager.cpp
        )
target_link_libraries(linkalibr_lib ${thirdparty_libraries})
target_include_directories(linkalibr_lib PUBLIC src)

##################################################
# Adding different executables
##################################################
add_executable(ros_test_node src/ros_linkalibr_test.cpp)
target_link_libraries(ros_test_node linkalibr_lib ${catkin_LIBRARIES})

add_executable(ros_calib_init src/ros_calib_init.cpp)
target_link_libraries(ros_calib_init linkalibr_lib ${catkin_LIBRARIES})

set(LIB_TBB_DIR /usr/lib/x86_64-linux-gnu)
add_library(libtbb SHARED IMPORTED)
set_target_properties(libtbb PROPERTIES IMPORTED_LOCATION ${LIB_TBB_DIR}/libtbb.so.2)

add_executable(ros_calib_init_optimizer src/ros_calib_init_optimizer.cpp)
target_link_libraries(ros_calib_init_optimizer
        gtsam
        ${catkin_LIBRARIES}
        libtbb)
```

ManChrys, 2023-09-06:
use @alwynmathew cmakelists.txt and if you have error about fstream just change in ros_calib_init_optimizer.cpp the fstream line with this 

`#include <fstream>
`

mahammadirfan, 2024-02-06:
@alwynmathew @songWell @SubMishMar @ManChrys @SyahirMuzni 
Hi,
I am still getting the error at the end...
[image]

Anyone solved this ?

jiheddachraoui, 2024-07-04:
hello @alwynmathew  i tried what you suggested but it didnt workn ihave the same error