unmannedlab/imu_lidar_calibration #20: Error in command

ManChrys, 2023-09-06:
Hello im trying to use the package but i have the follwoing error.

```
geo@geo-X570-GAMING-X:~/imu_lidar_calib_ws$ roslaunch linkalibr ros_calib_init.launch
WARNING: Package name "imuPacket" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits, underscores, and dashes.
... logging to /home/geo/.ros/log/8b8db6a8-4c96-11ee-9ed7-18c04d2d923b/roslaunch-geo-X570-GAMING-X-8624.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://geo-X570-GAMING-X:45593/

SUMMARY
========

CLEAR PARAMETERS
 * /ros_calib_init/
 * /ros_calib_init_optimizer/

PARAMETERS
 * /ros_calib_init/bag_durr: -1
 * /ros_calib_init/bag_start: 0.0
 * /ros_calib_init/ndt_resolution: 0.25
 * /ros_calib_init/path_bag: /home/geo/imu_lid...
 * /ros_calib_init/topic_imu: /imu/data
 * /ros_calib_init/topic_lidar: /livox/lidar
 * /ros_calib_init_optimizer/accelerometer_noise_density: 0.000726621235914
 * /ros_calib_init_optimizer/calibration_result_filename: /home/usl/catkin_...
 * /ros_calib_init_optimizer/gyroscope_noise_density: 0.000145127436329
 * /ros_calib_init_optimizer/max_frames: 550
 * /rosdistro: melodic
 * /rosversion: 1.14.13

NODES
  /
    ros_calib_init (linkalibr/ros_calib_init)
    ros_calib_init_optimizer (linkalibr/ros_calib_init_optimizer)

ROS_MASTER_URI=http://localhost:11311

WARNING: Package name "imuPacket" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits, underscores, and dashes.
process[ros_calib_init-1]: started with pid [8655]
process[ros_calib_init_optimizer-2]: started with pid [8656]
[ INFO] [1693995589.514415334]: Loaded accelerometer_noise_density: 0.000726621
[ INFO] [1693995589.515386647]: Loaded gyroscope_noise_density: 0.000145127
[ INFO] [1693995589.515639217]: Loaded max_frames: 550
[ INFO] [1693995589.515882047]: Loaded calibration_result_filename: /home/usl/catkin_ws/src/imu_lidar_calibration/linkalibr/data/I_T_L_init.txt
[ INFO] [1693995589.556121861]: topic_imu: /imu/data
[ INFO] [1693995589.556718272]: topic_lidar: /livox/lidar
[ INFO] [1693995589.556747568]: ndt_resolution: 0.25
[ INFO] [1693995589.557603021]: bag start: 0
[ INFO] [1693995589.557620484]: bag duration: -1
[ INFO] [1693995589.558055419]: ROS BAG PATH is: /home/geo/imu_lidar_calib_ws/calibr_test.bag
[ INFO] [1693995589.593069648]: Time start = 1.69395e+09
[ INFO] [1693995589.593097901]: Time end = 1.69395e+09
[ INFO] [1693995589.771248623]: Frame: 0 / 550
[ INFO] [1693995589.775224768]: Frame: 1 / 550
[ INFO] [1693995589.779356698]: Frame: 2 / 550
[ INFO] [1693995589.783978267]: Frame: 3 / 550
[ INFO] [1693995589.788576863]: Frame: 4 / 550
[ INFO] [1693995589.794416132]: Frame: 5 / 550
[ INFO] [1693995589.806269103]: Frame: 6 / 550
[ INFO] [1693995589.813131564]: Frame: 7 / 550
[ INFO] [1693995589.816918209]: Frame: 8 / 550
[ INFO] [1693995589.822207075]: Frame: 9 / 550
[ INFO] [1693995589.829455498]: Frame: 10 / 550
[ INFO] [1693995589.836961520]: Frame: 11 / 550
[ INFO] [1693995589.841952109]: Frame: 12 / 550
[ INFO] [1693995589.848563976]: Frame: 13 / 550
[ INFO] [1693995589.854470373]: Frame: 14 / 550
[... 353 lines cut ...]
[ INFO] [1693995598.154600834]: Frame: 368 / 550
[ INFO] [1693995598.154644977]: Frame: 369 / 550
[ INFO] [1693995598.202624076]: Frame: 370 / 550
[ INFO] [1693995598.246627261]: Frame: 371 / 550
[ INFO] [1693995598.248914330]: Frame: 372 / 550
[ INFO] [1693995598.254567788]: Frame: 373 / 550
[ INFO] [1693995598.294599637]: Frame: 374 / 550
[ INFO] [1693995598.338610225]: Frame: 375 / 550
[ INFO] [1693995598.382615904]: Frame: 376 / 550
[ INFO] [1693995598.391153985]: Frame: 377 / 550
================================================================================REQUIRED process [ros_calib_init-1] has died!
process has finished cleanly
log file: /home/geo/.ros/log/8b8db6a8-4c96-11ee-9ed7-18c04d2d923b/ros_calib_init-1*.log
Initiating shutdown!
================================================================================
[ros_calib_init_optimizer-2] killing on exit
[ros_calib_init-1] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete

```

the log file : [roslaunch-geo-X570-GAMING-X-8624.log](https://github.com/unmannedlab/imu_lidar_calibration/files/12536304/roslaunch-geo-X570-GAMING-X-8624.log)

Any help would be appreciated. Thanks in advance.

ManChrys, 2023-09-12:
@SubMishMar Hello my friend. Could you help me ?

As you mentioned on another issue https://github.com/unmannedlab/imu_lidar_calibration/issues/14#issuecomment-1280350300

They have both times as you can see below.

`geo@geo-X570-GAMING-X:~$ rostopic echo /livox/lidar -n 1`

```
header: 
  seq: 396
  stamp: 
    secs: 1694461327
    nsecs: 800928593
  frame_id: "livox_frame"
height: 1
width: 10000
fields: 
  - 
    name: "x"
    offset: 0
    datatype: 7
    count: 1
  - 
    name: "y"
    offset: 4
    datatype: 7
    count: 1
  - 
    name: "z"
    offset: 8
    datatype: 7
    count: 1
  - 
    name: "intensity"
    offset: 12
    datatype: 7
    count: 1
  - 
    name: "tag"
    offset: 16
    datatype: 2
    count: 1
  - 
    name: "line"
    offset: 17
    datatype: 2
    count: 1
is_bigendian: False
point_step: 18
row_step: 180000
```
`geo@geo-X570-GAMING-X:~$ rostopic echo /imu/data -n 1`

```
header: 
  seq: 12054
  stamp: 
    secs: 1694461328
    nsecs: 124748828
  frame_id: "imu_link"
orientation: 
  x: -0.00626751063573
  y: 0.0388246986508
  z: 0.0128333417534
  w: 0.999144046249
orientation_covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
angular_velocity: 
  x: 0.00864937994629
  y: 0.00250116037205
  z: -0.00423938082531
angular_velocity_covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
linear_acceleration: 
  x: -0.75243973732
  y: 0.0525619126856
  z: 9.85795879364
linear_acceleration_covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]

Any help would be appreciated! Thanks in advance.

BloodLemonS, 2025-04-21:
Have you solved it?
Can you tell me how to solve it?
Look forward to your reply

BloodLemonS, 2025-04-21:
I know it seems to be related to the recording time of the bag, but after running 550 frames, I still encountered the problem of process has died, only displaying Rot3, Euler Angles, and Marginal Cuvariance. What is the problem?